coseyfannitutti/discipline #6: Flashing bootloader with 6 pin IVR ASP programmer

liampelletier11, 2020-01-16:
I recently received the parts to assemble this keyboard but it wasn't as a kit, so I have to load the boot loader manually. I'm having trouble doing so even with the instructions and was hoping to get some help. I am running Windows 10 and the programmer I am using is linked below.

https://www.amazon.ca/USBtinyISP-Programmer-Bootloader-Download-Interface/dp/B07GYGC4NJ/ref=pd_rhf_se_p_img_13?_encoding=UTF8&psc=1&refRID=3255FZGS4NVRG2BWX159

coseyfannitutti, 2020-01-16:
Have you installed WinAVR?

Building on Windows:
You need WinAVR for the firmware, see http://winavr.sourceforge.net/.
To build the firmware with WinAVR, change into the "USBaspLoader" directory,
check whether you need to edit the files mentioned above (e.g. change device
settings, programmer hardware, clock rate etc.) and type "make" to compile
the complete source code.
After you upload the code to the device with "make flash", you should set
the fuses with "make fuse".

liampelletier11, 2020-01-16:
I have WinAVR and I'm assuming I build it from the command prompt, but whenever I use the make command I get error messages.
[image]

coseyfannitutti, 2020-01-16:
Just go ahead and run make flash and make fuse. Make is only really necessary if you need to change anything. Doesn't hurt anything to run it, but you'll get the errors you see there if nothing has changed since the last compile.

liampelletier11, 2020-01-16:
Now I get an error saying that it could not find USBtiny device

[image]

coseyfannitutti, 2020-01-16:
Not sure. That's an issue with your ISP programmer and Windows not detecting it. Perhaps a driver needs to be installed. Is it showing up in device manager? I've used the sparkfun pocket avr programmer (also a usbtiny device) as well as usbasp with this bootloader, and they required nothing more than installing WinAVR and running the commands.

Install zadig at https://zadig.akeo.ie/

Run. Click Options > List all devices.
Select your programmer from the drop down menu.
What driver is installed?

liampelletier11, 2020-01-16:
The driver that is installed is libusb0

[image]

coseyfannitutti, 2020-01-16:
You can try replacing it with libusb-win32.
Otherwise, I'm not sure what to tell you. Possibly a faulty device.
Maybe a restart is necessary after installing WinAVR.

If that still doesn't work I would recommend trying a USBasp instead. They are inexpensive.

If you decide to try a USBasp instead, a very simple change would need to be made to lines 40 and 41 in the Makefile.inc file.

1. Remove the hash at the beginning of the line "PROGRAMMER = -c usbasp"
2. Add a hash at the beginning of the line "# PROGRAMMER = -c usbtiny"

liampelletier11, 2020-01-16:
I was looking at other solutions and would it be possible to use an arduino uno to attempt and flash the atmega32a instead of the programmer directly?

coseyfannitutti, 2020-01-17:
I can really only speak to the functionality of the two programmers I’ve personally used, which are USBasp and Sparkfun Pocket AVR Programmer.

liampelletier11, 2020-01-17:
Ok so I tried some new things and now I get a new error with make flash and make fuse. Also, am I supposed to have the usb-c cable attached when I run the make commands?

[image]

coseyfannitutti, 2020-01-17:
If the power indicator led turns on when connected to the programmer than no you do should not have the usb cable plugged in. Are you sure that your microcontroller is in the correct orientation (notch to the left)?

jrgaitan, 2020-01-17:
Hi, I am having a similar issue, I'm using an USBasp, already changed the makefile.inc but in the following error shows up when I use the make and make flash command.
[image]

coseyfannitutti, 2020-01-17:
Ah okay I know this problem.

Download this file: http://www.madwizard.org/download/electronics/msys-1.0-vista64.zip
Unzip it and copy msys-1.0.dll to the \utils\bin directory of your WinAVR folder.

You may need to restart your computer.

jrgaitan, 2020-01-17:
It worked, thank you very much.

coseyfannitutti, 2020-01-17:
Glad you got it working.

liampelletier11, 2020-01-18:
I have the microcontroller in the correct orientation

coseyfannitutti, 2020-01-18:
Another thing to check is the orientation of your 6-pin cable on the header of the pcb. Make sure the notch [as seen in this picture](https://cdn.sparkfun.com/assets/learn_tutorials/2/9/2/2x3-labeled.jpg) is in the same orientation as the silkscreen on the pcb, notch pointing upward.

In the last screenshot you sent, your computer has recognized the programmer, but it is not communicating with the microcontroller. I can see WinAVR is installed, and you are entering the correct commands.

If you are certain that your pcb doesn't have any shorts, the microcontroller is in the correct orientation, and the cable is connected correctly to the 6-pin header, the issue most likely lies with the isp programmer or the 6-pin cable. The only advice I can give for that is to use one of the two programmers that I have previously suggested. I can't recommend using any others because I have not personally tried them.

Julian4314, 2020-01-18:
I need to chip in on this conversation, because i am running into the exact same problem liampelletier11 has. 
I am using an USBasp, changed the right parts of the makefile but can´t seem to connect to the board. But i was able to flash a qmk hex file onto an atmega328p with a basic bootloader using the same setup, so all connections seem to be working just fine.
It´s just that i can´t seemt to find a way to either flash the discipad or discipline board i have here.

coseyfannitutti, 2020-01-18:
I would double check everywhere on the board for shorts. There must be a physical issue causing that. Someone else in the conversation using a USBasp flashed the bootloader just fine. Works for me as well.

Julian4314, 2020-01-18:
It´s working now, but for what ever reason i had to flash another bootloader first before i was able to use the USBasp to flash your USBaspLoader.

coseyfannitutti, 2020-01-18:
That is bizarre. There shouldn’t be anything on a blank ATmega32A that would prevent it from being flashed with this bootloader.

liampelletier11, 2020-01-21:
If there were any shorts on the PCB that would be causing this error, which connectors would most likely be shorted? I'm pretty sure that all of my joints are ok but I have attached a picture as well just in case.

[image]

coseyfannitutti, 2020-01-21:
The USB port is generally the most likely spot, but yours looks pretty good.
Does the power indicator turn on when you connect the ISP programmer (USB not connected)?

liampelletier11, 2020-01-21:
The power LED does turn on with the ISP connected

coseyfannitutti, 2020-01-21:
My apologies, but I've really got nothing else then. This is not an issue with the bootloader. If everything else has been done properly, it's either the ISP programmer or the cable going from the programmer to the header on the board.

liampelletier11, 2020-01-22:
That's ok, thank you very much for your help! I've ordered a USBasp to try and see if that fixes the issue

liampelletier11, 2020-01-22:
Turns out it was the programmer not working, had no problems at all with the USBasp. Thanks once again for your help!

coseyfannitutti, 2020-01-23:
No problem at all. Glad to hear you got it worked out.

closed 2020-01-23 by coseyfannitutti

quantd55, 2020-03-16:
Does this programmer can flash bootloader for atmega32a? ty.
[image]

coseyfannitutti, 2020-03-16:
@quantd55 I have only personally used Sparkfun Pocket AVR Programmer and USBasp (with 10-pin to 6-pin adapter) with this repository. I'm sure there are others that may work, but I have not tested them. I cannot say with certainty which programmers will work outside of the two recommended above.

quantd55, 2020-03-17:
> USBasp

Ty, I will change to the USBasp.