coseyfannitutti/discipline #7: dxf file size when uploaded to ponoko

Ashtoruin, 2020-02-15:
[image]

I was looking to do a run myself for me and a friend in the UK and figured I'd start by looking at Ponoko (since you suggested it for the cases) but when I upload the file for the discipline guard (I checked the Mysterium Guard + Nav also) I noticed that the sizes are way off. I'm assuming they're just scaled wrong and I need to manually override the size but you don't actually list what the sizes should be. I can probably figure it out myself but it would be nice if we had some values to sanity check against.

Judging from the size of the boards:
Discipline Guard: 304.9W?
Mysterium Guard: 347.7W?
Mysterium Nav: ?

coseyfannitutti, 2020-02-18:
Manually entering 304.9 x 26.8 will get you the correct results.

There just seems to be issues with Ponoko's new auto detection system they recently launched.
The files were designed in mm and when measured in AutoCAD they show the correct size, and I'm not sure what would cause a scaling of one third, as mm to in would be either .04x or 25.4x.

Rest assured, though, if you manually enter those dimensions it will come out correct. I have be using them for a long time over several projects, and have never received an incorrectly sized piece.

Uploading it to other services shows it scaling correctly.
[image]

Ashtoruin, 2020-02-18:
Yeah, I actually opened them up in Fusion and double checked my sizes but I figured it might be nice to document this in the Readme or something for people wanting to print their own that aren't as familiar with CADD programs and such.

coseyfannitutti, 2020-02-18:
I agree. I want to keep the main page as simple as possible, but I have added a note about manual dimension entry for Ponoko in the Readme for the case folders of Discipline and Mysterium.

closed 2020-02-19 by coseyfannitutti