coseyfannitutti/discipline #8: Version with split backspace?

ElectricCoffee, 2020-03-20:
Hi, love the design, I'm just wondering if you could also make a version of the PCB that has a split backspace?

That might sound strange at first, but it frees up room for the <code>`~</code> key, so it doesn't have to share space with <code>Esc</code>.

For an example of this in action, see the Sun Model 7:

[image]

lstubbig, 2020-04-07:
A similar suggestion: I would love to see a version with split spacebars (2.75u 1.25u 2.25u). It's something really useful that practically no off-the-shelf keyboards offer. Would fit really well to the DIY character of the board and make it a definite buy for me.

ElectricCoffee, 2020-04-07:
@lstubbig totally agree, a split spacebar would be neat for enabling an easy-to-reach extra custom bindable key. Though I fear finding half spacebars might be hard...

lstubbig, 2020-04-07:
@ElectricCoffee Most keycap sets have an option for small spacebars nowadays (at least GMK and MT3 from Drop). You can also get relatively cheap XDA keycaps from Aliexpress.
@coseyfannitutti Would you be open to this? I don't have experience with KiCAD but if you give us some pointers maybe we can even add those features ourselves?

ElectricCoffee, 2020-04-10:
@lstubbig for the spit spacebar I'd instead have it split into four 1.5u keys or maybe three 2u keys.
Something about the Romeo's wide-narrow-wide spacebar cluster looks wrong to me...

mobin-amanzai, 2020-04-29:
@ElectricCoffee you could make a comprise sand share the left shift with the`~`` key

savyexe, 2020-07-30:
> @ElectricCoffee Most keycap sets have an option for small spacebars nowadays (at least GMK and MT3 from Drop). You can also get relatively cheap XDA keycaps from Aliexpress.
> @coseyfannitutti Would [...]

hey, i know it's been months since you guys talked about this but i wanted to say that if you do end up doing a 2.75u 1.25u 2.25u split spacebar version and @coseyfannitutti gives you permission to place it as a sort of groupbuy i'd be interested

closed 2020-08-25 by coseyfannitutti