coseyfannitutti/discipline #10: discipline-top-2.dxf file

cesaremiliodo, 2020-04-21:
Hey,

The new discipline-top-2.dxf file doesn't work on Ponoko. I try to upload it and the webpage crash.

ninizheng18, 2020-04-24:
I'm having the same problem with both of the top.dxf files. They either crash the page or they don't like to be uploaded at all.

discipline-top-1.dxf
discipline-top-2.dxf 

Not sure if this is an issue with the files or Ponoko?

coseyfannitutti, 2020-04-25:
Well, I have sort of a bizarre solution for you guys. The filename was actually the problem.

This is definitely an issue with Ponoko's system at the moment. I have previously used these files to have the case cut and have just looked them over extensively for any gaps and errors.

After seeing you guys saying this I tried and had the same issue.

I was able to get them to upload simply by removing the second dash from the filename, and they uploaded just fine for me now. Renaming them discipline-top1 and discipline-top2 was the only thing that changed.

No idea why having a second dash would now cause an issue when it previously didn't, but this should solve your issue. I have been able to upload it successfully several times.

ninizheng18, 2020-04-25:
It was still a little finicky even after I changed the names but it eventually worked. Thanks!

Is there any issue with using the "Clear impact modified acrylic" for the 2mm top piece instead of the regular clear acrylic? I figure, the more durable, the better.

coseyfannitutti, 2020-04-25:
@ninizheng18 No, you won’t have any problems with that at all. It’s just a little more expensive.

closed 2020-04-25 by cesaremiliodo