coseyfannitutti/discipline #11: Question about ISO support

ilyen85, 2020-04-27:
Hey Coseyfannitutti

First of all I would like to thank you for your fantastic work!

So my question, I missed the group buy on Friday, it was sold out when I managed to enter my details :( So I figured I will buy / create the necessary parts.

I have never did anything like this before so I have a couple of questions.
If I use a pcb manufacturing service, do I need anything else than the files in the gerber folder in this repo?

With these files will I be able to have an ISO compatible pcb?

Thank you again!

mozaic-dev, 2020-04-29:
@ilyen85 yeah, there is even a picture of the white PCB showing off ISO enter

coseyfannitutti, 2020-04-29:
Currently, no the ISO support and other changes for V2 are not yet in the repository. The version here only has ANSI compatibility.

I will be opening a larger pre-sale on Friday at 12PM US Eastern time to help meet some of the demand after they sold out so quickly last week.

I will be updating this repository with the ISO compatibility and other changes some time after that pre-sale has run.

ilyen85, 2020-04-30:
Thank you! I will wait for that, I hope i will be able to grab a kit (or two) :)

Since I know that ISO compatibility is not yet in the repo I will close this issue

closed 2020-04-30 by ilyen85