coseyfannitutti/discipline #13: Bluetooth for V3

louia, 2020-05-06:
is there any plans to add bluetooth and battery ?

coseyfannitutti, 2020-05-06:
Not at this time, no. That would require either using an expensive 32u4
Feather BLE from Sparkfun or adding smd parts that also need to be
programmed in such a way that would prevent this kit from being sold and
built as it is now.

The lack of extensive Bluetooth support with qmk and potential headaches
with connectivity limitations for users doesn't help. It's just not a
priority at the moment.

However, this design could be edited by someone looking for a project to
use something like the 32u4 Feather BLE from Sparkfun. That board will,
however have significantly less i/o pins and would require a port expander
and significant changes to the firmware. One of my other smaller boards
like the Romeo or Discipad could be adapted for use with the Feather
without using a port expander.

louia, 2020-05-06:
Thanks for your explication and your response

louia, 2020-05-07:
After some search on the www, i discover [this project](https://github.com/Galzai/MK32) but it doesnt compatible with QMK and on reddit i found this reply if one day it may be usefull for you or for someone else : [reddit](https://www.reddit.com/r/MechanicalKeyboards/comments/9d6w1y/prototyping_boards_for_my_esp32_based_wireless/e5lqt2p?utm_source=share&utm_medium=web2x)

coseyfannitutti, 2020-05-08:
Maybe. We'll see how porting of new microcontrollers and alternate firmware options develop. If it is possible to keep the user experience as easy and friendly as QMK and VIA, then it may be something to consider.

closed 2020-05-09 by coseyfannitutti