coseyfannitutti/discipline #14: Kit Plate and Foam

goranzarkovic, 2020-05-08:
Does the .dxf plate file for the Discipline that is in the /case folder also work when not using the acrylic case from Ponoko? I was thinking of using that .dxf file to get an aluminum plates at 1.6mm.

Also are there maybe .dxf files for the foam as well?

goranzarkovic, 2020-05-13:
resolved.

closed 2020-05-13 by goranzarkovic

Elkasitu, 2020-06-20:
@goranzarkovic did you find the .dxf? I'm also looking for a backplate that I can print myself

jesusignazio, 2020-06-21:
@Elkasitu I think it is provided here https://github.com/coseyfannitutti/discipline/tree/master/case

Elkasitu, 2020-06-21:
@jesusignazio From checking the files, the bottom case model seems to be for the full-acrylic case and is at least 15mm taller than the PCB.

The kit backplate seems to be the same size, the holes align with the PCB plus comes with extra holes for feet, here's a pic
[image]

@coseyfannitutti is there any way you could share the file for the backplate included in the kit?

coseyfannitutti, 2020-06-21:
The bottom is already in the repository. In the gerber folder there are files for the bottom.

Elkasitu, 2020-06-22:
Damn thanks, don't know how I missed that