coseyfannitutti/discipline #16: LED backlit keys

darthrookie, 2020-05-22:
Is there a plan for a future version with LED backlit keys? That would be IMO all this keyboard is missing. If I find the time I could edit the schematic and perhaps the PCB routing, but since I am new with keyboards, that could take some time since I am not familiar with common components and pin outs for this application.

Also perhaps a version with a dedicated "~" key. For a solution right now I was thinking on moving the ESC key to where DEL is, assign DEL as a secondary function of BACK, and put ~ where ESC is.

dhanakane, 2020-09-17:
Most people will reassign ESC to CAPS LOCK.

SuperNici, 2021-05-08:
As @dhanakane mentioned this can be done in [VIA](https://caniusevia.com/)