coseyfannitutti/discipline #18: Plate layout?

andywarburton, 2020-06-02:
Hey. I'm self building the Discipline V2 board (because a) I'm impatient and b) I missed the groupbuys) and was wondering if you could please make the gerbers for the switch plate available? Cheers!

coseyfannitutti, 2020-06-02:
This repository is currently V1. It will be updated to V2 sometime after pre-orders have been fulfilled. Both versions of the board can be built plateless and will be functionally the same if using PCB mount (5-pin) switches.

andywarburton, 2020-06-02:
Ah cool. That makes things a little easier. I guess I'll just keep chugging away with my V1 build and build a V2 afterwards :)

andywarburton, 2020-06-02:
btw... thanks so much for being so open (source) with your designs. It's so refreshing to see someone just openly give away their designs for free like this. I promise I'll get in on a future groupbuy if our timelines ever cross :)

coseyfannitutti, 2020-06-03:
I did just remember that 65% tray mount style plates such as those that are sold by kbdfans and kprepublic are compatible with Discipline.

And no problem regarding the open sourcing. Customization and open source hardware were what got me into keyboards in the first place. Happy to contribute in the same manner and give people the satisfaction of building something from start to finish or provide a jumping off point to redesign for their own needs.

andywarburton, 2020-06-03:
Does this look like the right one? (it looks like they have it in carbon fiber and brass too) 
https://www.aliexpress.com/item/32999460415.html

coseyfannitutti, 2020-06-03:
Yep! That should work just fine. I’ve seen several users build theirs with both carbon fiber and brass.

coseyfannitutti, 2020-06-03:
Just make sure when building to attach screws and standoffs BEFORE soldering the switches. The screws will not be accessible once the plate is there and switches are soldered.

coseyfannitutti, 2020-06-04:
> 
> 
> This repository is currently V1. It will be updated to V2 sometime after pre-orders have been fulfilled.

coseyfannitutti, 2020-06-26:
V2 source and plate files published.

closed 2020-06-26 by coseyfannitutti

andywarburton, 2020-06-27:
AMAZING!