coseyfannitutti/discipline #23: What PCB thickness is recommended?

jesusignazio, 2020-06-29:
[image]

dhanakane, 2020-06-29:
1.6mm should be fine, but 1.2mm makes it a little bit easier to solder. I
would think a slightly thicker plate would help with stiffness if you are
not going to use a switchplate too.

coseyfannitutti, 2020-06-29:
You need to do 1.6mm. Do not do 1.2mm. Screw in stabilizers are intended to be used with 1.6mm PCBs. You would need to make inserts for the stabilizers to use a 1.2mm PCB.

closed 2020-06-29 by coseyfannitutti

kiwisaurio, 2020-07-02:
hey, I'm trying to print it on jlcpcb but it's my first time printing a pcb and I don't know the size of this pcb, can you post a screenshoot of your pcb setting you placed on jclpcb?

thank you

coseyfannitutti, 2020-07-02:
That is just their gerber viewer timing out. Typically, you can go to File
Manager on JLC and find it there with the gerber preview and click "Quote".

Otherwise you can manually enter the dimensions, which are 305 x 123 mm.

kiwisaurio, 2020-07-02:
ok, I just realised what you told me, now I got the question, how many layers should I put? Is this configuration ok?
[image]

dhanakane, 2020-07-05:
Just a quick follow up to the PCB thickness question and screw in stabilisers. As a first timer, I notice that the stabilisers come with spacers. Would simply adding an additional spacer of 0.4mm or even cutout electrical tape where the screws attach not fix the problem of thinner PCBs? I ask as I have ordered some PCBs at 1.2mm and don’t want them to go to waste. I am also planning on using them with a metal plate for added stiffness, so PCB thickness should hopefully not affect the typing experience too much.

jesusignazio, 2020-08-04:
@coseyfannitutti what about the plate thickness? Is it 1.6mm too? Thanks

coseyfannitutti, 2020-08-04:
@dhanakane You can use [these shims designed by ai03)[https://github.com/ai03-2725/StabilizerShims] to use screw in stabilizers with a 1.2mm PCB. The spacers that come with C3 stabilizers will not be sufficient for this purpose. Those are simply for protecting from shorts if the screw is going in an area near a trace.

@jesusignazio Yes, plate thickness should be 1.6mm as well for switches to be able to clip in properly.