coseyfannitutti/discipline #24: JLC PCB right configuration

kiwisaurio, 2020-07-02:
After searching through the internet a lot, I didn't see anyone posting how to properly print the PCB, so I want to ask if this configuration is ok.

Thank you.
[image]

coseyfannitutti, 2020-07-02:
It should be changed to the default 1.6mm for the PCB to work properly with
screw in stabilizers. 1.2mm can only be used if you make .4mm inserts for
your stabilizers, which for this particular build is not worth the trouble.

I would also recommend selecting "Yes" for the "Remove Order Number"
option. If "No" is selected JLC will silkscreen an order number onto the
PCB. Unless a specific location is specified they may place it in an area
that may be visible and unsightly on your completed build.

kiwisaurio, 2020-07-03:
> It should be changed to the default 1.6mm for the PCB to work properly with screw in stabilizers. 1.2mm can only be used if you make .4mm inserts for your stabilizers, which for this particular [...]

Perfect, really appreciate it, if the remaining settings are ok, the build is complete for me. Really thankful ^^

closed 2020-07-03 by kiwisaurio