coseyfannitutti/discipline #26: spacing problem

andywarburton, 2020-07-13:
Just a heads up, I tried ordering the plate file from JLCPCB and they rejected it on the basis of:

"the spacing between slot to slot or outline require 4mm at least in case of break off , kindly replace with new file ."

I'll try a few things and see if I can figure out where the problem lies and will update here if I find anything out.

coseyfannitutti, 2020-07-13:
The edge is exactly 3mm from the cutouts because they've told me multiple
times in the past that it needs to be 3mm. The problem lies with whoever
reviewed that file and made up that new number, not the file.

I've had them cut nearly 2000 plates with that spacing with no issues,
though occasionally they will add an extra charge.

You could email them saying you're okay with the risk. You could also try
cancelling and resubmitting to get a different reviewer, which has been
successful for a few people that I've spoken to that have had that issue.
Otherwise, I'd recommend just using someone else to cut the plates.

andywarburton, 2020-07-13:
ha ha... funnily, thats pretty much what I ended up doing. I made some tiny tweaks and resubmitted.

andywarburton, 2020-07-14:
You were right, the second reviewer approved and then bumped the price slightly!

closed 2020-07-14 by andywarburton