coseyfannitutti/discipline #28: Issues loading complete files for FR4 plate on pcbway and jlcpcb

IOPNR, 2020-07-15:
Loading .zip files from the plate folder in master files but I'm receiving error messages.

Per jlc "Cannot Identify Board Outline"
Per pcbway "There is no drill file and silkscreen layer."

Am I missing something?

coseyfannitutti, 2020-07-15:
You just need to manually enter the dimensions, 305.9 x 96.4mm. Rounding the numbers is fine, or you can copy and paste those numbers from here. That's just for the instant quote. The true dimensions will be taken from the gerber files.

"There is no drill file and silkscreen layer." - More of a warning than an error. Most PCBs will have drills or silkscreen markings, but this one doesn't. There is nothing on the plate to drill or silkscreen, so that's fine.

IOPNR, 2020-07-16:
Sounds great, thank you!