coseyfannitutti/discipline #29: [Feature request] Is there any ways to solder a 2811 led strip into the pcb, then control it by controller

coadotvn, 2020-07-16:
I just solder it into the power led holes, then control it by a RF remote.
If there is any way to solder a led strip, to be controlled by the pcb, it will be best.

coadotvn, 2020-07-16:
[image]

IOPNR, 2020-07-19:
That's awesome. What kind of led strip did you use if you don't mind my asking?

coadotvn, 2020-07-21:
> That's awesome. What kind of led strip did you use if you don't mind my asking?

any 5v led strip will works, just solder its pin to power led holes on the pcb, it will works

nhanldtn, 2020-08-27:
You can solder the ws2812's data pin on PIN D0 and define config with QMK firmware. I did it and worked fine

stevenvo, 2020-09-09:
> You can solder the ws2812's data pin on PIN D0 and define config with QMK firmware. I did it and worked fine

Can you pls share where is pin D0?

nhanldtn, 2020-09-10:
you can find it from datasheet. This is 14th pin (RXD -PD0)
[image]