coseyfannitutti/discipline #30: Issues loading dxf files into ponoko

hollowsmart, 2020-07-17:
Attempting to upload any of the dxf files on ponoko's website leads to various errors such and invalid files and timeout errors.

coseyfannitutti, 2020-07-17:
Did you download the entire repository or did you try to download the files individually?

Downloading the files individually will give you text/html files, which would lead to errors like that.

Otherwise, if you are using the files from the full repository, the issue is just with Ponoko's uploader. These are the exact files that I have used many times. I have noticed with Ponoko's uploader that if files are taking an extremely long time to upload, for some reason changing the file name can help. I don't know the reason why that is. If the issue persists you can just email them the files. They are very helpful and easy to work with.

hollowsmart, 2020-07-17:
I downloaded the files individually. Just tried uploading from the master repo files and they worked. Thanks for pointing that out!

closed 2020-07-17 by hollowsmart