coseyfannitutti/discipline #34: Origin/license for the USB-C connector symbol/footprint

piit79, 2020-08-15:
Hi @coseyfannitutti, could you please let me know what's the origin of the USB4085-GF-A USB-C connector KiCad schematic symbol and footprint? I would replaced the Mini USB connector on the [Plaid](https://github.com/hsgw/plaid) ([PR](https://github.com/hsgw/plaid/pull/27)) with this connector, but I'm not sure it's OK to use the schematic symbol/footprint and commit it into the Plaid repo,

Thanks a lot!

coseyfannitutti, 2020-08-24:
The USB4085-GF-A footprint that I use is a modified version of the one that was in ai03's USB Type-C library. The symbol I just made myself because at the time his library didn't include one, and I wanted to be sure that the pins would be assigned correctly.

It is absolutely fine for you to use them for Plaid.

piit79, 2020-08-24:
Thanks, much appreciated!

closed 2020-08-24 by piit79