coseyfannitutti/discipline #37: DXF Acrylic case error

CptGerry, 2020-08-23:
When trying to open the .DXF files in Adobe Illustrator I get an unknown file error, I tried to download and clone the entire repository, individual files and rename them and they still cannot be viewed

coseyfannitutti, 2020-08-23:
I've personally never worked with .dxf files in Illustrator. These files were created using Fusion 360 and saved as 2018 AutoCAD DXF and can be opened and edited with Fusion 360, AutoCAD, LibreCAD, and FreeCAD. I'm sure there are others, but I can personally confirm all of the files work fine with those.

CptGerry, 2020-08-23:
Thanks, I opened them with FreeCAD and re-exported them as DXF and they now open without problems in Illustrator :)

closed 2020-08-23 by CptGerry