coseyfannitutti/discipline #40: SA Profile keycap binding to bottom corner of case

dhanakane, 2020-09-15:
I was wondering if it was possible to amend the layer of the case (closed layer) so there is a bit more clearance for SA profile keycaps. I have this issue with both the right arrow key and the left CTRL key. I could sand the layer down, but it would be nice to not have to do this.

coseyfannitutti, 2020-09-15:
Are you certain that you have the 2x "discipline-top-2" pieces in the correct orientation?

On those two pieces two of the inner corners are sharper, which are meant to be on the bottom side to allow for keycap clearance.

dhanakane-tw, 2020-09-16:
Mea culpa. That was exactly the problem I had. Thanks for taking a look and giving me a solution.

I wonder if it's possible to use the same type of layer for that specific "shape" unless there is a structural/construction reason to not use the same design for the similar looking layers.

dhanakane, 2020-09-17:
That was exactly it, thanks! Is it not possible to have the same corners for all the closed layers to make assembly simpler?

ph-One, 2020-12-21:
Did the exact same thing. I spent a good bit looking everything over before I realized the subtle difference in the corners. I'm not saying any changes are needed bc I should have looked closer, but just +1'ing that I too dealt with this, and if it was more obvious it would have saved me some time