coseyfannitutti/discipline #43: Some BOM parts OOS

pand06, 2021-01-07:
There are a few items on the BOM list that are OOS and I was wondering what if there are alternatives or other version of them that would work.

SuperNici, 2021-05-02:
Which ones are you talking about?

SuperNici, 2021-05-06:
What a coincidence! I ordered on mouser today aswell! 

Because i didnt want to wait till june 7 i decided to go with [this](https://www.mouser.ch/ProductDetail/667-ECE-A1HKA4R7) capacitator which will hopefully work the same. 

Also make sure you got the right ic which would be [this one](https://www.mouser.ch/ProductDetail/556-ATMEGA32A-PU). Untill yesterday it was wrong in the BOM so make sure you ordered the right one.