coseyfannitutti/discipline #45: Personal build information/help

debo, 2021-04-01:
Hi @coseyfannitutti I'm writing to you because, as per subject I'd like to build the Discipline v2 by myself but by being completely new to the hobby, I'm not quite sure where to go to get the parts manufactured, specifically the pcb/plates and the acrylic case.

If my understanding is correct, based on the online stream I saw with MechMerlin you usually use JLC PCB for prototyping and Elecrow for the final product, is that correct? In a normal scenario both the bottom and top plate are made of PCB material? Just to understand if I can get them made all by Elecrow.

About the acrylic instead, I understand from the documentation that you use Ponoko instead. Is that correct too?

One more question, I notice in the photos that there are various engraving on the components are those part of the gerber and dfx files already or do I need to modify something else?

Last but not least, when I use JLC PCB/Elecrow, are there settings on their tool that I have to change other than the colour of the PCB?

Thanks a lot for your time and help.

SuperNici, 2021-05-08:
Im currently in the process of building this keyboards as well, I feel like I understand the process well enough to answer some of your questions.

> you usually use JLC PCB for prototyping and Elecrow for the final product

Prototyping is not required unless you modify the PCB yourself and since coseyfannitutti's design has been thoroughly tested you can skip the Prototyping step all together and order straight from JLCPCB or Elecrow, although most people i know ordered from JLCPCB.

debo, 2021-05-09:
Thanks a lot @SuperNici this was super super helpful, really appreciated.

robopuff, 2021-06-25:
Hey guys, I am buildong it for myself as well and I've used JLCPCB and it looks and works well. It was not easy for me to get proper M2 screws and standoffs since no store would sell one with correct size and esthetically pleasing - aliexpress worked fine for it. With that I've also ordered steel plate and it is a blast to use! 

Now I need to cut just the plexi and was wondering if you know correct thickness @SuperNici for the kitfoot?

fourstepper, 2021-11-06:
@SuperNici would you care to elaborate how you handled the top plate? Looking at my order, the size doesn't get correctly detected by jlcpcb. Any ideas? Thank you a lot :)

robopuff, 2021-11-06:
You can get the dimensions of the top plate from laser cut files, can't remember what they are right now but I've used it to cut in steel plate and it fits perfectly

fourstepper, 2021-11-06:
Sorry for my ignorance, I am still quite new to this. Which files are the laser cut files? Thank you a lot

robopuff, 2021-11-06:
> Sorry for my ignorance, I am still quite new to this. Which files are the laser cut files? Thank you a lot

No worries, I had a million questions when I've built mine 😅

You can check it here https://github.com/coseyfannitutti/discipline/blob/master/plate/discipline-plate-lasercutting.dxf

fourstepper, 2021-11-06:
Thank you, that worked!

SuperNici, 2021-11-06:
Personally I made my discipline plateless, see [my reddit post ](https://www.reddit.com/r/MechanicalKeyboards/comments/qmvntb/comment/hjc6ruf/?utm_source=share&utm_medium=web2x&context=3). In hindsight I regret not having ordered plates, but it works just fine without one.

The file you imported was the Gerber file, the missing dimensions should be the same as for the PCB itself, in this case:

`122.81 * 304.89`

This is an issue with the generated Gerber file, I tried fixing it and had no luck. I'll open an issue in just a second.

PS. @fourstepper you just replied while I was writing this comment, i'll still post it if others have the same issue :)

robopuff, 2021-11-06:
> Personally I made my discipline plateless, see [my reddit post ](https://www.reddit.com/r/MechanicalKeyboards/comments/qmvntb/comment/hjc6ruf/?utm_source=share&utm_medium=web2x&context=3). In [...]

Isn't PCB taller than the plate? I mean it's not covering the electronic components but rather just the keys

SuperNici, 2021-11-06:
Oh I totally forgot about that, my bad. It would probably still arrive fine, you'd just pay more... (JLCPCB would just waste a bunch of extra Fr4 Material since the keyboard plate itself gets cut out of the specified size)

What measurements did @fourstepper use? I'll update my comment accordingly.

fourstepper, 2021-11-06:
I used the measurements from the acrylic top plate as suggested by @robopuff, which reports 305.7 x 96.2 mm. Seems like that could work?

SuperNici, 2021-11-06:
Thanks, that looks about right, I updated [the issue I opened](https://github.com/coseyfannitutti/discipline/issues/60) ( ´ ▽ ` )b

robopuff, 2021-11-06:
> 
> 
> I used the measurements from the acrylic top plate as suggested by @robopuff, which reports 305.7 x 96.2 mm. Seems like that could work?

I've just checked against what service provider I've used sent back to me and indeed it was 305x96mm JLCPCB have their checks done before accepting a job so maybe they would take a look into it as well before misprinting/cutting/etc things 😅

fourstepper, 2021-11-06:
@robopuff I have a lot of time and have some screws and standoffs already, as I am also in the process of building a CRKBD. Thank you for the recommendation though!

SuperNici, 2021-11-07:
> I've just checked against what service provider I've used sent back to me and indeed it was 305x96mm JLCPCB have their checks done before accepting a job so maybe they would take a look into it as well before misprinting/cutting/etc things 😅

@robopuff i dont think it should matter too much as long as it is big enough, the exact cutting measurements are taken from the gerber file after all so it really should not be a problem

NiciTheNici, 2023-05-27:
Heyall, just out of curiosity... How many of you ended up building this keyboard? Got any pics?

Heres mine!
[image]

robopuff, 2023-05-27:
Daily driven since built, use U4T switches with steel plate, it's a joy to use. 

[image]