coseyfannitutti/discipline #46: USB-C connector detaching

TheStockie, 2021-04-06:
I've been having problems with the USB-C cable tearing the connector off the PCB, I was wondering if people have encountered this problem and what you've done to prevent this.

Thanks a lot for the help!

wiebel, 2021-05-03:
Have you soldered on the larger tabs? I can't imagin the connector gfetting detached without breaking the pcb. You have to apply liberal amount of heat (high wattage and/or longe heating time) to those tabs and the connected ground plane due to it's high thermal mass to get the solder flowing. But if done properly  you shouldn't be able to simply "tear" it out.

coseyfannitutti, 2021-05-05:
The four feet on the bottom if fully soldered should be more than enough to hold it in place and would require quite a bit of force to break. It's possible that if you did solder those four feet that perhaps they were cold joints that weren't fully soldered.

closed 2021-05-05 by coseyfannitutti