coseyfannitutti/discipline #51: Screw type not clear

SuperNici, 2021-05-09:
The type of screws needed is not listed anywhere besides in the [build guide PDF](https://static1.squarespace.com/static/5c533d33348cd92b886e544d/t/5f4597da4c8f740579aec0ca/1598396382120/DISCIPLINE+BUILD+GUIDE.pdf) (page 4) 

I do not understand what type of screw is meant by ***6x M2 10+3mm standoffs***
I would think it means ***6x m2x10mm screws*** and ***6x 3mm standoffs*** but I am not sure.

[image]

Armastardo, 2021-05-16:
That's the correct name. The standoff you're looking for is the one in your picture:
[image]

omniphx, 2021-05-20:
10+3mm standoff = male-female
6mm standoff = female-female

SuperNici, 2021-05-20:
Thank you, this helps a lot!

closed 2021-05-20 by SuperNici