coseyfannitutti/discipline #52: Help with Bootloader/ Flashing

mishnoc, 2021-05-27:
Hello, I recently acquired some discipline 65 pcbs and parts and have soldered one together. I am now at the point where i need to flash the micro controllers. However I have no clue how to go about this. I have a usbasb programmer (https://www.amazon.co.uk/gp/product/B06XYV162N/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1) and have downloader winavr but after that I am completly lost. I have no experiance at all with this kind of thing. Any help would be greatly appreciated.

khail-k, 2021-05-27:
Does this help?
https://static1.squarespace.com/static/5c533d33348cd92b886e544d/t/5e9fd1d198a4166c9ccc5b68/1587532241535/DISCIPLINE+VIA+GUIDE.pdf

alkeeb, 2021-05-27:
Do these help?

Bootloader:
https://github.com/coseyfannitutti/discipline/tree/master/doc/bootloader

Flashing VIA guide:
https://static1.squarespace.com/static/5c533d33348cd92b886e544d/t/5e9fd1d198a4166c9ccc5b68/1587532241535/DISCIPLINE+VIA+GUIDE.pdf

mishnoc, 2021-05-28:
Thanks for the reply. Im fairly sure the micro controllers are blank. I tried to follow the bootloader instructions but honestly have no idea what I'm doing.

jesusignazio, 2021-05-29:
> Thanks for the reply. Im fairly sure the micro controllers are blank. I tried to follow the bootloader instructions but honestly have no idea what I'm doing.

Hi. 

In this issue you have info. https://github.com/coseyfannitutti/discipline/issues/6 and in the guide too.
The only point is that you have to change the line 41 in the Makefile.inc file to 'PROGRAMMER = -c usbasp' if you are going to use an usbasp flasher.

@alkeeb you need special hardware in order to flash the bootloader if you have bought a new microcontroller. Once you have a bootloader, you will be able to use the programming buttons of the keyboard.

alkeeb, 2021-05-29:
@jesusignazio I have spent some time looking into this and yes I would need to flash a bootloader. Maybe you can help answer some questions I have:

- Can I flash USBaspLoader with an arduino or is it limited to the arduino bootloader; if so will the arduino one work for this project?

- AVR or ISP programmer, does it matter? I have not found any good descriptions on this, just that each one supports a different chip. I know that coseyfannitutti used an AVR.

- What are the pins on the PCB for? 
[image]
Can I connect the programmer to this or do I need a breadboard circuit like in all the tutorials.

jesusignazio, 2021-05-29:
> @jesusignazio I have spent some time looking into this and yes I would need to flash a bootloader. Maybe you can help answer some questions I have:
>
> * Can I flash USBaspLoader with an arduino or is [...]

This is how I flashed it:

–I used an usbasp programmer, the same you linked in the OP.
–I connected the usbasp to the keyboard (in the three pins you asked) and to the USB from PC aswell.
–The software stuff: Download avrdude, clone the usbasploader repository (is important to clone the correct branch). I have read that there is a common problem with drivers in Windows, but I use Linux so I can't help you in this.
–Modify the line in Makefile.inc file to 'usbasp' in order to make the usbasp work.
–Run make and make install commands in avrdude.
–Once the bootloader is flashed, the usbasp won't be necessary anymore.
–Connect the keyboard to the PC with an usb keyboard and use the two buttons from the keyboard to flash VIA.

alkeeb, 2021-05-30:
That clears up a lot, thank you.

mishnoc, 2021-05-31:
After some trial and error I finally managed to get my controller flashed. Huge thanks to both of you for the help <3.

closed 2021-06-02 by mishnoc