coseyfannitutti/discipline #53: Alternative Part

cozmiz, 2021-05-30:
Hi, I would like to ask if there is any alternative part. Some of the components are out of stock and will only be in stock in ten or more weeks. The part in question in the 4.7uF Capacitor (UMT1H4R7MDD) and the 16mhz crystal (ABL-16.000MHz-B4Y). I was only able to get my hand on the PCB v1 and the backplate.

jesusignazio, 2021-05-30:
- https://es.aliexpress.com/item/1005002075527957.html
- https://es.aliexpress.com/item/1005002562834681.html

cozmiz, 2021-05-30:
would this work https://www.digikey.com/en/products/detail/abracon-llc/ABL-16-000MHZ-B2/675236 for the 16mhz crystal?
and this https://www.mouser.com/ProductDetail/667-ECE-A1HKA4R7 for the 4.7uf Capacitor?
I am not sure if I want to wait for 30-50 days for the components to arrive via aliexpress.

mishnoc, 2021-05-31:
I bought some parts recently and used this, https://www.mouser.ie/ProductDetail/Panasonic/ECE-A1HKS4R7?qs=rMMd5vBiahogzu%252BjM7Ehlw%3D%3D. Similar to what you linked but is the same size as the one in the BOM. It worked fine for me.

closed 2021-06-01 by cozmiz