coseyfannitutti/discipline #54: ISO compatible optional acrylic case

Gizzzle, 2021-06-08:
Hi @coseyfannitutti ,

any chance you can provide a ISO compatible version of the optional bigger acrylic case?

willi-dev, 2021-08-21:
> Hi @coseyfannitutti ,
> 
> any chance you can provide a ISO compatible version of the optional bigger acrylic case?

hi, @Gizzzle , I have made my own version (based on the design in this repo) for ISO compatible with modified some of the thickness of some parts. Available in svg & dxf format.
you can download it here: https://drive.google.com/drive/folders/1NyenxRh25v5X8mUzmUIyBVmncQgihrfy?usp=sharing
I hope it can helps

Gizzzle, 2021-08-23:
> hi, @Gizzzle , I have made my own version (based on the design in this repo) for ISO compatible with modified some of the thickness of some parts. Available in svg & dxf format.
> you can download it [...]

Hey Willi,

thanks for your reply. Am I right to assume that each file needs to get manufactured one time?

Best regards
Alex

willi-dev, 2021-08-24:
> Hey Willi,
>
> thanks for your reply. Am I right to assume that each file needs to get manufactured one time?
>
> Best regards
> Alex

Hey Alex,
yes, each file only needs to be manufactured one time.