coseyfannitutti/discipline #57: Discipline V2 Groupbuy

charles0917, 2021-07-17:
Hi a lot of people have not uet received their discipline V2 from last November's groupbuy. Can you please give an update whether they will be shipped? Last update you gave was from over three months ago. Thanks.

charles0917, 2021-07-28:
@coseyfannitutti So I'm assuming you just decided to keep the money and not give the products? Pretty sure that decision will hurt you down the road as you've broken the law.

coseyfannitutti, 2021-07-29:
That's absolutely not my intention at all. About half the orders have already gone out, and a lot (hopefully the remainder of all outstanding orders) will be going out in the next week.

I know this has been a rough rough group buy to say the least. Several unforeseen delays and trying to do it all myself got the best of me on this one. I understand that everyone that's contributed to this project works hard for their money, and moving forward I intend to make a more pointed effort to respect the fact that you've spent your hard earned dollars at the store and also bring in some help with packing and sorting.

zanderc1, 2021-07-30:
Hey man, really hope you’re doing well during these tough times. A lot of people are still checking on the website for updates. Are you planning on making an update about this groupbuy on that website soon? I thought it would help with some people out with their concerns.

ningyuuu, 2021-07-31:
@coseyfannitutti thanks for checking in! I was a little worried you were facing some health issues during this period, but glad to hear everything is well. Will patiently wait for the order to deliver.

tifanrepo, 2021-08-01:
> @coseyfannitutti Hey man, really hope you’re doing well during these tough times. A lot of people are still checking on the website for updates. Are you planning on making an update about this groupbuy on that website soon? I thought it would help with some people out with their concerns.
> 
> (EDIT): Theres a on-going subreddit about this topic and a lot of people wanted an update about the groupbuy from the website and a lot of people are still worried about their orders and they (seem) that they want an update about it in the website.

Same thing here. Many people are wondering what its going since there have been no updates for months now. Could you please give a general statement?

thank you

marc-fez, 2021-08-03:
> That's absolutely not my intention at all. About half the orders have already gone out, and a lot (hopefully the remainder of all outstanding orders) will be going out in the next week.
> 
> I know this has been a rough group buy to say the least. Several unforeseen delays and trying to do it all myself got the best of me on this one. I understand that everyone that's contributed to this project works hard for their money, and moving forward I intend to make a more pointed effort to respect the fact that you've spent your hard earned dollars at the store and also bring in some help with packing and sorting.

@coseyfannitutti much appreciated, thanks for the update. I know times are tough and stuff happens but if you could toss a note on the pre-order updates page once a month or so most of us would be cool with it. We get that long lead times and delays are part of this hobby. It's the total silence that is concerning.

charles0917, 2021-08-18:
So another 3 weeks has passed, has anyone gotten any updates?

bean-ders, 2021-08-18:
> 
> 
> So another 3 weeks has passed, has anyone gotten any updates?

No, I sent an email 3 weeks ago as well after viewing this thread. If I could just get a response, I do not care if this takes another 6 months - as long as this comes in at some point and is still being tracked.

tifanrepo, 2021-08-18:
With this radio silence is hard to believe everything is going ok. I have decided to do a credit card chargeback. This situation has shown everything but respect for the customer. I mean, We are discussing this on a GITHUB ISSUE, for gods sake.

maxim-sm, 2021-08-18:
@coseyfannitutti I don't want to come off as rude but I really feel that I need to say this. I really don't understand why it is so hard to write a quick update on the main website. You wrote, "...and moving forward I intend to make a more pointed effort to respect the fact that you've spent your hard earned dollars at the store...", but you still refuse to update the website after several requests, although it will clearly show the respect you're talking about.

Please take a look at these two Reddit threads, [here](https://www.reddit.com/r/MechanicalKeyboardsUK/comments/oc9go2/has_anyone_in_the_uk_received_their_discipline/) and [here](https://www.reddit.com/r/MechanicalKeyboards/comments/m3mldw/cftkb_discipline_v2/). Please read the confusion and frustration of the people who bought the kit that could've been avoided with a few short updates.
Moreover, on the page for the Discipline, you state that "Delays are possible, but you will be notified in a transparent manner if any such delays occur". No update since April is not, in any way, to be notified a transparent manner.

And I'm not even mentioning all the people that emailed you for support with the kit or help with lost packages, and didn't get any response for several months. Or the issues that are opened in this repository without an answer.

Again, I'm sorry if all of this sounds rude or aggressive, but this situation is really frustrating. At this point I just hope to receive the kit without any issues because I know I will wait another half an year for a reply.

zanderc1, 2021-08-20:
@coseyfannitutti Hey man, trying to check in with you. Is everything okay and stuff? I don’t wanna come off rude but some haven’t received an update since 3 weeks ago when you said it was gonna ship the week after. And most people who does not know about this GitHub thread are starting to think that you ran off with the money. Any update will help your customers a lot, even if its a simple “Sorry about the very quiet 3 months as I was working hard to get these orders out..etc.” That would also help more younger customers who purchased this with a visa gift card from their hard earned cash, and the people who still have hope in you that you would send out all of the orders.

We REALLY don’t care how much you would take to get the orders out. Hey, it might take 10 months more. We just need updates about it so we know where our product is and if our hard earned cash isn’t getting stolen.

Really hope the you would read this, make and update and stuff. Or don’t. We can’t control you. But keep in mind, a lot of people are getting frustrated.

bean-ders, 2021-08-28:
> 
> 
> I hate bumping thread on GitHub, but another week has passed here without an update. No replies to mails, no shipping notification, nothing on the website. The only reason why I didn't file a claim with my bank is because it's already past the time limit to do so. Please give an update... It takes literally less than 5 minutes to post a sign of life and maybe 5 more to give an honest timeline instead of "next week" and then radio silence for another month...
> 
> And in all honesty I don't even want or need the kit anymore. If a refund is easier, feel free. My order number is 02916.

I receive mine without any shipping notice/email response out of the blue. 

I agree, radio silence is the worst way to deal with this. Being open and honest gets you far in the community. Most reasonable people will understand delays, especially with current supply shortages and shipping backups.

d-daemon, 2021-09-13:
@coseyfannitutti Can you please provide an update on the website on your status? I've sent you multiple e-mails about updating my shipping address with no response. Would allowing us to cancel our orders and getting a refund help alleviate your own personal workload? Any update would be greatly appreciated as the community is quite upset over this GB. Any communications is better than no communication, thanks.

hygrogen, 2021-09-28:
I sent an inquiry every month, but I didn't get any answer, and I thought I was still acting when I saw a post on github at the end of July, but I'm very frustrated to see that no notice has been posted and that people around me haven't been shipping the goods. Please, just one word is okay, so write down the objective current situation and post it. You stated, "We inform you transparently about the delay." Keep what you said. 
I'm very angry, but I don't think I've given up because nothing has changed.

d-daemon, 2021-09-28:
I requested a refund via e-mail on Sep 25th and it was processed the next day for anyone else interested in a refund as most banks probably won't charge back due to how old the order is.

zanderc1, 2021-09-28:
> I requested a refund via e-mail on Sep 25th and it was processed the next day for anyone else interested in a refund as most banks probably won't charge back due to how old the order is.

I heard that he’s shipping orders that were ordered at the ~23 November of last year this month

d-daemon, 2021-09-28:
> > I requested a refund via e-mail on Sep 25th and it was processed the next day for anyone else interested in a refund as most banks probably won't charge back due to how old the order is.
> 
> I heard that he’s shipping orders that were ordered at the ~23 November of last year this month

I ordered before November 23rd and I've seen some reddit threads where people who've randomly got their order with no tracking update or response to their address update who had order numbers way after mine, where they needed to fetch the package from their old address. I moved and Bryan didn't reply to any of my address update requests. I don't want to stress about the entire logistics process anymore and to be honest, I fell out of interest with the board due to the brand's (CFTKB) handling of the delays/communication.

hygrogen, 2021-09-28:
I placed the order on November 17th, but why is the person who ordered on November 20th receiving a delivery notification, and why am I still pending?
I'm about to lose my mind.

zanderc1, 2021-09-28:
> I placed the order on November 17th, but why is the person who ordered on November 20th receiving a delivery notification, and why am I still pending?
> I'm about to lose my mind.

I don’t know, im not sure.

curdl, 2021-10-04:
Came here from the reddit threads unfortunately as my #02930 order still hasn’t arrived. Sending emails hasn’t worked so I guess I’ll try GitHub 😞

hydrogen1-1, 2021-10-04:
> Came here from the reddit threads unfortunately as my #2930 order still hasn’t arrived. Sending emails hasn’t worked so I guess I’ll try GitHub 😞

My status is still pending. My order number is #2906. I'm so angry

lhoste, 2021-10-31:
Order #02971, placed on November 17, 2020 at 12:06 PM ET. I am located in France, and have yet to receive any update whatsoever about this groupbuy. If anyone in France has received one, let me know.

EuthaNasi, 2021-11-04:
> Order #2971, placed on November 17, 2020 at 12:06 PM ET. I am located in France, and have yet to receive any update whatsoever about this groupbuy. If anyone in France has received one, let me know.

Order #02840, also placed on the 17th of November and located in The Netherlands. No shipping update received. Only communication was about my adress change on which they responded after a month or so. Still waiting on 2 kits

SimRicard, 2021-11-08:
I'm in the UK and I emailed asking if Bryan would ship to a US-based parcel forwarding service and he responded and has refunded me the difference in shipping. I have a tracking number now but it has yet to actually ship out. Just an idea for international people who still don't have their boards.

sunnyhwang, 2021-11-10:
From NZ, order low 3000, haven't heard anything, sent him an email for shipping address update and haven't received any reply at all. Gonna issue a refund

QLaille, 2021-11-12:
Heya, From France, order #02867 placed on November 17, 2020 at 2:29 AM ET, and I have had no update at all since the one posted on the website in April :/

NGEddie, 2021-11-13:
> I'm in the UK and I emailed asking if Bryan would ship to a US-based parcel forwarding service and he responded and has refunded me the difference in shipping. I have a tracking number now but it has yet to actually ship out. Just an idea for international people who still don't have their boards.

I am in the same position, UK based and not heard anything at all and he hasn’t responded to multiple emails.  I too asked him send it to a US forwarding company, but again no response. I would also be willing to accept a different colour in case that is an issue. Order number 3077.

I don’t see we have any other options other than start to explore legal options.

rustygloves, 2021-11-23:
@NGEddie if you go the legal route please feel to include me. At this point I just want my money back and I do not wish to support this project.

zanderc1, 2021-11-28:
bro wheres our kits D: @coseyfannitutti

SimRicard, 2021-11-29:
> I am in the same position, UK based and not heard anything at all and he hasn’t responded to multiple emails. I too asked him send it to a US forwarding company, but again no response. I would also [...]

Well I've had no updates since I posted my reply, it has now been a month since I got the tracking number and it's still on "shipping label created, USPS awaiting item". I've tried emailing Bryan for an update but, predictably, I've not received a response.

bizander20, 2021-11-29:
Hello,

I am collecting data on this order. 

I have two surveys:

Requires google sign on, no email or personal data collected: https://docs.google.com/forms/d/e/1FAIpQLSf2QRRDXitGTe67Yn3Wz5ohPX4zKTJiSwch2kgwHCh6d09loQ/viewform?usp=sf_link

One that does not require google sign in: https://docs.google.com/forms/d/e/1FAIpQLSenXBQuPjMng4SLacK7SA5NoIHcKgGQ1sbCChaZZve0P747Bw/viewform?usp=sf_link

I will be posting the data and some visualizations on Reddit but wanted to give the GitHub posters time to input their experience with CFTKB. 

Hope we all get our kits - or refunds - or even a response.. 

Thanks for your input on this. Hopefully we can visualize with data just how bad this group buy  has gone, and the community will put some pressure on CFTKB.

sahsuvar, 2021-12-14:
@coseyfannitutti is there any update?

zanderc1, 2021-12-14:
Hey guys, just wanting to let you guys know I got my kit, shipping notification 2 weeks ago and arrived yesterday. Order 03290, Canada based. I hope you guys get all of your kits.

lhoste, 2021-12-14:
Thanks for sharing, i am getting a bit more hope for my kit to arrive now.
> Hey guys, just wanting to let you guys know I got my kit, shipping notification 2 weeks ago and arrived yesterday. Order 03290, Canada based. I hope you guys get all of your kits. 
> 
> 
> 
> This groupbuy has been really hectic, and im just relieved that I got my kit after a year of waiting. Just wished that it had more communication between the seller and the buyer, and we would get updated at a transparent matter (as promised).
> 
> 
> 
> I share feelings with other people in a sense that we didn’t get an update since july and some (who haven’t seen this thread) from april.
> 
> 
> 
> anyways I hope you guys get your kits, because this is truly a nice board, and I would hope that cftkb ships all of them.
> 
>

knorr3, 2022-01-23:
Order No. 02868 here.
Wrote them a mail about any updates last year. Got no response.
The order is still "pending" and hasn't been shipped.

refundrequest, 2022-01-31:
Hello @coseyfannitutti. I'd like to request a refund for Order No. 02913 from the Philippines. I have lost interest in the product due to the lack of updates in the website or replies to my few emails. Thank you.

thinkbulecount2, 2022-02-17:
Hi, I'd like to request a refund for Order No. 02610 from Canada. I got a tracking number late last year, but I followed up with DHL and said the shipper never shipped it. Please refund me. Thanks.

liger45, 2022-03-04:
I ordered two Discipline65 kits but never got a notification email at all. Emailed them a few times and never got a response back. How hard is it to just give us an update on what's happening? I mean its pretty common in this hobby to have delays and long wait times, but not getting any updates on shipping status and not replying emails is kind of weird...

hydrogen1-1, 2022-03-10:
My order number is #02906 and I don't understand how long it takes to pack and ship only one a day.
There was no announcement, and looking down on the contact through e-mail calls, 
As time goes by, I think this guy is a scammer.
I ordered with high expectations in South Korea, but only endless waiting remained.
I've sent you a refund request email several times, but no answer comes back.
Are the other orderers in the same situation as me?

ZebastianBjorkqvist, 2022-03-14:
My order number is #028XX for 1 kit and I haven't received any emails about my order or a response on emails I've sent. The lack of communication has been very disappointing and i assume this was a scam. @coseyfannitutti please respond to us and just give an update if you plan to send all the missing orders.

indeed808, 2022-03-14:
Damn man, what the hell. NO UPDATES, NO RESPONSE. Definitely a scammer. What are you, dead?

SimRicard, 2022-03-14:
> Well I've had no updates since I posted my reply, it has now been a month since I got the tracking number and it's still on "shipping label created, USPS awaiting item". I've tried emailing Bryan [...]

For what it's worth, I did eventually get my board. I know that people have had inconsistent results asking him to send to a mail forwarder but it's worth a go.

sahsuvar, 2022-04-11:
Did anyone from Europe get any updates? 5 Mails are unanswered. No Auswertung to refund, forwarding or any Status. Seems like it was a scam.

lhoste, 2022-04-11:
> Did anyone from Europe get any updates? 5 Mails are unanswered. No answer to refund, forwarding or any Status. Seems like it was a scam.

i still don't think it's a scam, but a badly handled group buy. Many things happened between the order and where we are at now. When i ordered we were laughing at the covid saying it was just a cold, and just see how big of deal it became. Also, chip shortages, stopped factory have been stopped and this has affected component supply for sure. 

While the lack of update is really not a good sign, we should still not be hard on bryan for the delay, if it is *only* delay.

We might even get the record of the longest group buy for some of us. 

Bit of an after though of writing the message, i think no one in the EU got the board, and it might be related by shipping difficulties. I wonder if an eu proxy could have speed up the process.

alanho, 2022-04-12:
my recommendation for @coseyfannitutti is: not giving an update will just get you deeper into a vicious cycle, more pressure and more stress. we all understand this is a difficult time and i'm sure people will understand, and people will try to help if you can share with us the difficulty you're having

chriswade, 2022-06-16:
Hey @coseyfannitutti, I have attempted to contact you via email for the last 12 months regarding my order (03228). I have moved house during this time and am concerned that my board has been lost in transit.

Please respond to this comment or emails at your earliest convenience.

goobzilla, 2022-08-04:
Hi @coseyfannitutti  i've been attempting to contact you via email for over a year now regarding my two orders (orders #02611 and #02912). I have recieved no replies whatsoever. I have been requesting a refund on the two orders since October of 2021. This is getting ridicious. Please refund me.

chriswade, 2022-08-04:
Hey @coseyfannitutti, it's been over a year since you said
>About half the orders have already gone out, and a lot (hopefully the remainder of all outstanding orders) will be going out in the next week.

Just wondering how things are going up there on Miller's planet?

kkurtson, 2022-09-27:
> Hey, @coseyfannitutti, are there any news? Has anyone received their order recently?

Order from November 2020, no answers on my emails for a refund - this thing is dead - will never order from this creator again

charles0917, 2022-09-27:
Guys it's time to move on. It's almost 100% at this point that he just robbed your 100ish dollars and ran. Pretty sure he was delibrately not replying to messages because the logo of his website changed late last year, so someone is still maintaining it.

kkurtson, 2022-09-27:
> Guys it's time to move on. It's almost 100% at this point that he just robbed your 100ish dollars and ran. Pretty sure he was delibrately not replying to messages because the logo of his website changed late last year, so someone is still maintaining it. "Difficult time" my as* 🤣

yup time to write it off as an experience and move on

zanderc1, 2022-09-27:
> 🤣
Im the lucky few that actually got their order and honestly it was just out of nowhere that I got my kit. I honestly would’ve thought that he has at lease shipped all of the kits by now- we did order them 2 years ago. I just hate how silent cosey has been through this whole ordeal and just not providing needed updates of the groupbuy

Then again its been 2 years since this groupbuy opened and the mk community has died down a slight bit so at this point im not surprised if people gave up on the gb

Well that was an expensive lesson for some of us, a $100 lesson for sure 💀

lhoste, 2022-11-09:
> If anyone is still following this issue and hasn't received thier keyboard. Yesterday I got a "back in stock" email for mysterium V2. So, it seems he's back in action and selling stuff. But atleast I'm still missing my discipline V2 kit.

Incredible news if he commit to it. Most of us in this thread are waiting for some form of communication.

kkurtson, 2022-11-09:
maybe one can write a warning in the mysterium github?

SimRicard, 2022-11-09:
I'm keeping an eye out for any posts advertising the sale on the mechanical keyboards or mechmarket subreddit, definitely not going to let this pass by without comment.

knorr3, 2022-11-09:
It's sold out again on the website.

itisibecky, 2023-03-06:
Anyone else still not received their order? I've sent an absurd amount of emails at this point, receive no response, no order, nothing.

itisibecky, 2023-03-06:
@coseyfannitutti Any update? It's been three years... at this point just refund me!!

goobzilla, 2023-03-06:
> Anyone else still not received their order? I've sent an absurd amount of emails at this point, receive no response, no order, nothing.

Nope, never recieved my order or any response to any of the emails I have sent. I have sent 17 emails at this point from May 2021 to date.

All I want is a refund....

@coseyfannitutti cmon dude.

j-dabrowski, 2023-05-14:
I also never received my board or any responses to email queries
Does anyone know what legal action is possible from Australia?

[19 more replies]