coseyfannitutti/discipline #62: Suggestion how to make discipline 65 more available to people outside of the USA

jovansijacki, 2021-12-19:
When I saw your design I must admit I fell in love.
But since I live in a country outside of the EU, the parts you recommended are scarce, expensive, and hard to find. I managed to find almost all parts on aliexpress, but one major problem persists.
I can't find the USB Type C connector part, the only option that even ships to my country is from mouser but the shipping is 10times the amount of the part (40USD shipping fee for a 4 dollar part).
So I am asking you is it possible to make an alternative design with a more mainstream connector board like this one I found on aliexpress for example.
( https://www.aliexpress.com/item/1005001300974530.html?spm=a2g0o.cart.0.0.3a7a3c00jCdf6k&mp=1 ) 
It could also look cool if it was mounted on some pins.
I know this is a long shot but I had to try.
Thanks in advance,
Jovan

jesusignazio, 2021-12-19:
https://github.com/coseyfannitutti/discipline/issues/36

jovansijacki, 2021-12-19:
Does the one you linked from aliexpress work?
I am currently on fiver looking for people who could modify the design, but this Is a cheaper option