coseyfannitutti/discipline #67: Connection issues on macOS

steakscience, 2022-05-09:
I'm encountering some connection issues when connecting a Discipline to a Mac.

At the login screen, the keyboard will work. After logging in, the keyboard will stop working and needs to be reconnected. I can reliably reproduce this 100% of the time.

This issue does not happen with other keyboards. It also does not happen with this keyboard on other OSes.

I was wondering if anyone else has had these issues or knows how to fix them. Thanks!

BenRoe, 2022-05-09:
Never had a similar problem.
Do you use a special firmware, or bootloader?
Try different usb port (without hub) and cables.

steakscience, 2022-05-09:
> Never had a similar problem.
> Do you use a special firmware, or bootloader?
> Try different usb port (without hub) and cables.

I have tried both: 
- USB C <-> USB C 
- USB A <-> USB C hub

I am (probably) using the latest VIA version, flashed a few months ago

BenRoe, 2022-05-09:
Try vial. It‘s open source, actively developed and has more features.
https://get.vial.today/

Still a strange behavior. Don’t know what it could be.

dhanakane, 2022-05-09:
Does hitting the reset button force the keyboard to register? I had a similar issue when switching inputs using a hub, and hitting reset when switching inputs fixed it.

steakscience, 2022-05-12:
> Does hitting the reset button force the keyboard to register? I had a similar issue when switching inputs using a hub, and hitting reset when switching inputs fixed it.

Before I try this, RESET will not reset the configuration right?

dhanakane, 2022-05-12:
No. You need to flash the keymap to do that and that involves putting the keyboard into bootloader mode.

steakscience, 2022-05-19:
> No. You need to flash the keymap to do that and that involves putting the keyboard into bootloader mode.
> […](#)

I've verified that pressing Reset after losing connection doesn't do anything

dhanakane, 2022-05-19:
Are you using VIA or something else? AFAIK, hitting reset forces the board to re-register.

If you’re using VIA and reset isn’t forcing the board to re-register then I’m out of ideas unfortunately. Maybe check the previous issues reported if there’s a fix?

NiciTheNici, 2023-07-10:
Did you manage to figure this out?