coseyfannitutti/discipline #70: Discipline inspired DIY Keyboard Troubleshooting

Steven2k6, 2022-12-26:
Hez, I designed and built a Keyboard, but I have some issues with the usbasp bootloader. When I connect the Keyboard to my pc, it says "Device Descriptor Request Failed" in the device-manager. I don't have the two zener diodes on my keyboard, are they the problem?
[image]
I use 3.3 instead of 5 volts, because I found out, that the keyboard doesn't work with 5 volts.

NiciTheNici, 2023-07-10:
Heya, did you manage to fix this? I'm mildly interesting in designing my own keyboard in the future too and this seems rather important :)

Steven2k6, 2023-07-10:
> Heya, did you manage to fix this? I'm mildly interesting in designing my own keyboard in the future too and this seems rather important :)

There where two 3.6V diodes missing on the data lines.

closed 2023-08-22 by Steven2k6