coseyfannitutti/discipline #72: cant see the build guide website is expired, but fortunately wayback machine exist

dhupee, 2023-08-05:
apparently, he hasn't renewed his Squarespace subscription, but a simple wayback machine does exist, and you can check this out

I hope it helps if you are having a problem in this regard

https://web.archive.org/web/20221205071237/https://static1.squarespace.com/static/5c533d33348cd92b886e544d/t/5f4597da4c8f740579aec0ca/1598396382120/DISCIPLINE+BUILD+GUIDE.pdf

MesterLum, 2023-08-08:
Thanks a lot!

NiciTheNici, 2023-08-29:
Oh man this really makes me wanna fork this awesome keyboard. Thanks for the issue!

dhupee, 2023-08-29:
Idk bout you but it might be a good idea to take every guide, rewrite it in markdown then save it here.

Imo it should remove the headache like this

hildebkg, 2024-02-03:
I'm a bit late to the party, but for convenience, I saved all the PDFs referenced in the [doc folder](https://github.com/coseyfannitutti/discipline/tree/master/doc) and submitted a PR: https://github.com/coseyfannitutti/discipline/pull/74/commits/10b8b82c8e8f274797e661fed40291813787a5ce